barebaric/rayforge #7: Marlin v2.1.1+ dialect: G moves with S

WildRikku, 2025-08-03:
I noticed you implemented GCode dialects and that's great! I'd like to suggest one that has become possible by newer Marlin versions and is used by my setup.

Since Marlin 2.1.1, S for power can be included with G0 and G1 (usually G1 since G0 is for travel moves). That would remove the need to turn the laser on and off with M commands during the operation, however, it is not defined if G0 and G1 share the value (meaning G0 would be with laser on when S was set on G1) or not. So it might be sensible to do `G0 Xx Yy S0 ... G1 Xx Yy Ss`.

The way the Creality software does it is different though and I am not entirely sure why: It uses `G1 S3` to turn the laser down before every G0 travel move. So their code looks like:

```
G92 X0 Y0 Z0
G90
G0 F3000
G1 F1500
M3 I
G0 X87.999 Y55.555
G1 S3
G0 X78.111
G1 X77.999 S501
G1 X42.333 S999
G1 X42.222 S501
G0 X32.333
G1 X32.222 Y55.555 S3
```

I'm not entirely sure what's going on there and will investigate, however, I wanted to point out the variant with S in the G command since I think that's currently not possible with the template you created, because that's one more variable to replace.

knipknap, 2025-08-03:
I see, thank you for the issue and for looking into this! I think it wouldn't be hard to add support for this in the dialect objects. The GcodeEncoder would need some changes to support it, but I think it is mostly trivial changes.

WildRikku, 2025-09-28:
I needed my printer for a bunch of 3D printing, but I'll have another laser project the next couple days and I'll take a look and give you feedback!

WildRikku, 2025-10-06:
Just to let you know: Due to my weird setup, I'm waiting for the 0.23 release to come to PyPi, which has the fixes for installation via pip. As soon as that drops, I'll run some tests. Thank you for your patience and effort.

knipknap, 2025-11-15:
I just added a dialect editor, too, so now there is full flexibility to customize gcode.

[image]

closed 2025-11-15 by knipknap

WildRikku, 2025-11-23:
Sorry for the delayed response. The built-in Marlin dialect does not include the power with G1, so that's not what I need, but the custom dialect editor should solve that and I think that's an even better solution. Looking forward to the release with that functionality.

knipknap, 2025-11-23:
Ah, ok. The editor was already released with 0.25 yesterday.

WildRikku, 2025-11-23:
Awesome, so it should arrive on Flathub soon. Thanks!