barebaric/rayforge #11: Cannot deselect

xupwup, 2025-08-10:
**Describe the bug**
Import the same DXF file twice. Try to move the first one, then the second one.

**To Reproduce**
Steps to reproduce the behavior:
1. Import dxf 1
2. move it
3. import dxf 1 again
4. try to move it
5. observe that you can only move the first and second one together

**Expected behavior**
deselect when nothing is clicked. Or when escape is pressed maybe.

**Versions:**
4f63a1e6

xupwup, 2025-08-10:
This looks like a promising project by the way!

knipknap, 2025-08-10:
Weird, I cannot reproduce it. And "deselect when nothing is clicked" should already happen...

knipknap, 2025-08-10:
Could it be that the object you imported has an almost transparent background, so clicking it would seem like clicking empty space but in reality you still clicked the object?

xupwup, 2025-08-11:
I can't reproduce it anymore either. No idea what happened

closed 2025-08-11 by xupwup