barebaric/rayforge #9: Group/ungroup feature

xupwup, 2025-08-10:
**Is your feature request related to a problem? Please describe.**
After importing a DXF, I want to move parts of the drawing. My DXF is made by onshape and contains auto-layouted components. I want to move things to optimally use my material.

**Describe the solution you'd like**
A group/ungroup feature. In this case mainly ungroup (the entire dxf file seems to be one group). I also expected to find this in the right-click menu, but there doesn't seem to be one.

**Describe alternatives you've considered**
you could theoretically split the dxf into pieces in another tool

**Additional context**
Add any other context or screenshots about the feature request here.

knipknap, 2025-08-11:
I thought about it, and it's even more complicated 😄 . Because resizing a group of already-rotated workpieces requires a new type of transformation (shearing/matrix transformation) for each workpiece. So this affects pretty much the entire application from import to generation to output. Implementation plan:

1. Add support for matrix transformations and for shearing single workpieces in the UI.
2. Change core data model to support groups: Introduce DocItem which contains attributes shared between Layer, Workpiece, and WorkpieceGroup. This involves changing the ops generator pipeline to support groups as well.
3. Change the canvas and UI to support grouping/ungrouping.
4. Change the importer.

I will try to make it part of release 0.20.

knipknap, 2025-08-19:
Released with Rayforge 0.20. Please re-open if you have any issues with it.