barebaric/rayforge #13: Migrate potrace to a version that doesn't depend on native libraries

matoushybl, 2025-08-21:
**Is your feature request related to a problem? Please describe.**
I tried running rayforge on Arch, but stumbled upon the dependency potrace requiring the agg native library to be present on the system, this library was only available in AUR, but even with it installed, rayforge couldn't start because of missing symbols.

**Describe the solution you'd like**
I think it would be beneficial to move away from this library. There are multiple benefits - less dependencies and support for multiple platforms - like running the software on a RPi or other hardware with different architectures. There is an alternative in [this](https://github.com/tatarize/potrace) library which seems to implement the same API only using numpy, but even this library doesn't seem to be actively maintained. 

**Describe alternatives you've considered**
* Use a different bitmap tracing solution

**Additional context**
I should be able to contribute a change like this myself, but I wanted to present this idea first before doing it. I understand that Arch support is not a requirement for this project, but I believe there are other more clear benefits of moving.

What do you think?

knipknap, 2025-08-21:
That would be nice! The problem with the pure python potrace is that it is slow... extremely so. Maybe using it as a fallback? I have not found any other tracers that come close in quality and speed to the one I used in the end.

I originally even started out with my own tracer, but while the basics are easy, it gets nontrivial when you want it to succeed on edge cases ...

matoushybl, 2025-08-22:
That makes sense, I thought all of the heavy lifting was done by numpy, but it doesn't seem so.
I also found [this](https://github.com/visioncortex/vtracer) in one of potrace's PRs, it is a Rust library (more easily manageable when using natively) with python bindings. It even directly compares to AGG and potrace. I'll try using it an report back the results.

knipknap, 2025-09-24:
So the app has changed so much since I made this branch that I re-did it based on latest master. It seems to work decently well, perhaps it also benefited from a couple of denoiser and image importer improvements that I made recently.

I am tempted to make the switch to vtracer, also because of #29 @cejote. But it would help if someone else also tested it...

fllmn, 2025-09-25:
Not a problem, right now I'm targeting release 0.22 for the flatpak build. If there's new release before that's done I can just drop the potrace dependency.

knipknap, 2025-09-29:
Merged with 305fee72a3c75df3a827c3dbc2504af2ce293f65

closed 2025-09-29 by knipknap

matoushybl, 2025-09-30:
Sorry for the inactivity and broken promises.

Thanks for the amazing work on the app! Looking forward to using my laser again after a lot of frustration with lightburn and meerkat.