barebaric/rayforge #15: Feature request: Support for micrometric resolution and scaled G-code workflow

LorenzoTell, 2025-08-28:
Hello,

I would like to share some work I did on Rayforge related to high-resolution micrometric engraving and propose a potential feature improvement. I made a Direct Laser Lithography System and this software comes handy in the workflow of all the open-source tools developed in LabµFab (Arg)

I managed to unpack the snapd package using a custom script on my machine (Lenovo, Intel Core i3-2310M, 7.6 GB RAM, Intel HD Graphics 3000, running Manjaro Linux with KDE Plasma 6.3.6, Qt 6.9.1, kernel 5.15.189). After unpacking, I modified the script to allow the print head in X and Y axes to work with a step size of 0.001 mm. Additionally, I modified the G-code generation to work with a scaled visualization, making it easier to design micrometric engravings on screen. The G-code is then rescaled automatically for the actual engraving. Finally, the script repackages everything back into a snapd and now I use the software with this mods.

It would be ideal if this functionality—a range of work unit scales—could be added as a feature in the preferences, so users can directly set micrometric or custom scales without needing to unpack and modify the snap manually.

I was unable to set up the development environment on my machine due to issues with pypotrace and potrace. That's why I unpacked the Snapd Install.

The software is a great and a measured response to a big necessity. I want to sincerely thank Barebaric for developing this software. There was nothing similar available for Linux, and my personal attempts to create something alike over the past two months only led to headaches and failed tries. I will be following the development of the program closely, and when possible, I would love to donate to support the excellent work you’ve done.

knipknap, 2025-08-28:
Wow, that sounds like an awesome set of features! Anywhere I can look at the changes you made?

I will try to provide an easier way to get a development environment running as well.

LorenzoTell, 2025-08-28:
I felt a bit embarrassed to push a branch with such rough hardcoding, but I can definitely share the modifications.

gcode.py with the mods: https://github.com/LorenzoTell/Misselaneous/blob/debdde04c3d3793268563b734d779adec51ccc79/gcode.py 

In laser_preferences_page.py I changed the "lower" value from 0.01 to 0.001:

spot_size_x_adjustment = Gtk.Adjustment(
    lower=0.001,
    upper=0.2,
    step_increment=0.01,
    page_increment=0.05,
)
spot_size_y_adjustment = Gtk.Adjustment(
    lower=0.001,
    upper=0.2,
    step_increment=0.01,
    page_increment=0.05,
)

This enable me to work in microscale with x10 scale svg images, and when gcode is generating I return them to original size. Thats because script importer, maybe how cairo is being implemented, lower the import resolution of the images (lower resolution for me is not being able to import a 30µm feature)

knipknap, 2025-09-10:
I just pushed support for this. The machine settings now have a new setting for the precision under "Advanced":

[image]

You can test by upgrading from the Snap edge channel.
Let me know if you have any issues with it.