barebaric/rayforge #24: Embedded generators

klattimer, 2025-09-18:
There are already python libraries for generating cool laser cutting things, e.g. https://github.com/florianfesti/boxes and https://github.com/lkesteloot/jigsaw 

It seems that it would be fairly simple to create, at least at first, a basic UI for accessing the functions of libraries like this to provide an attractive feature set.

knipknap, 2025-09-18:
Thanks for the feature request! I think this would be cool to have. I was thinking that things like this would be especially great to support the creation of material libraries in the app.

I am not sure yet if I would add this to the app directly, or if I would implement addon support first...

knipknap, 2026-01-06:
Rayforge just gained a package manager and plugin support (#34), so this could now be implemented as an addon.

Docs: https://rayforge.org/docs/latest/developer/plugin-docs.html