barebaric/rayforge #31: ARM architecture target (RaspberryPi, etc)

srsabu, 2025-09-24:
Is anyone else working on attempting to get Rayforge to run under Raspbian (or similar ARM based Linux)?  I've started to look at it, and I'm currently learning what I need to about Pixi but wanted to see if there were opportunities to collaborate or at least not replicate something someone else was doing.

knipknap, 2025-09-24:
I am not aware of anyone working on it. One thing that maybe would make this easier is if #13 were done, as potrace and libagg are the biggest headaches to get running...

fllmn, 2025-09-25:
I'm at least intending to figure out if the flatpak builds/runs on aarch64 as part of #28  as it needs to included in the flatpak meta data.

I'll let you know how it goes

fllmn, 2025-09-26:
Good news, it does build under emulation (qemu) although it takes ages. Runs better than expected on a RPi 3 but I won't be able to test cutting right new

srsabu, 2025-09-27:
I'm kicking off a flatpak build (on the Pi 4).  We'll see how long it takes.

srsabu, 2025-10-02:
So the build has died a few times on the Pi 4 (always on OpenCV).  I am getting a heatsink/fan for it to see if it's just throttling to un-usability based on the prolonged load.

srsabu, 2025-11-20:
It runs well enough that I'd say close this, and any bugs can be created as a new issue.  Thanks!

closed 2025-11-20 by srsabu