barebaric/rayforge #35: Allow Baud rate to be edited manually -OR- update to add 1,000,000

loneregister, 2025-09-27:
**Is your feature request related to a problem? Please describe.**
I am unable to connect via USB to my MonPort Mega 70W laser. The documentation from MonPort specifies that the baud rate should be 1,000,000.  I am able to manually set this in lightburn, which then works.

**Describe the solution you'd like**
Add 1,000,000 to the drop down (quick). Or allow manual entry of the baud rates. Or allow drop down usage + manual entry of baud rate.

**Describe alternatives you've considered**
I am not aware of an alternative.

**Additional context**
None

knipknap, 2025-09-27:
I assume that this was done by #36, can I close it?

loneregister, 2025-09-27:
Resolved by 36.  Closing

closed 2025-09-27 by loneregister