barebaric/rayforge #38: Missing dependencies in pip package

WildRikku, 2025-09-29:
**Describe the bug**
Regression of #5 

After upgrading from 0.17 to 0.22, I needed to manually install via pip: `pycairo PyGObject blinker numpy pyvips`

Then rayforge still crashed ` OSError: cannot load library 'libvips.so.42': libvips.so.42: ` so I guess there's also some new dependencies that need system-wide installation.

This is on Manjaro Linux (Arch-based).

WildRikku, 2025-09-29:
Is it possible that the dependencies do not populate to PyPi? I downloaded the .tar.gz from PyPi and can't find the requirements in there.

Since I installed those dependencies one by one and got a different error message every time and they were all indeed not installed, I'm pretty sure the issue is that these dependencies do not get installed when installing with pip.

knipknap, 2025-09-29:
I think I found the issue! Should push a fix soon.